hackndev/zinc #275: stm32f2, aka Cortex M3, aka Pebble Watch!

photex, 2015-01-30:
Hello,

I just ordered a pebble (http://getpebble.com) and I'm interested in using Rust with their C SDK and toolchain. The chip in the watch is the stm32f2 which is an ARM Cortex-M3. I see in platform.yml that there are other chips listed from ST. My question is what's the best way for me to try and add support for this in zinc? The wiki doesn't have a guide for that. Should I just start posting questions on the mailing-list?

Cheers!

bharrisau, 2015-01-30:
Zinc is really for bare-metal work (e.g. writing the pebble operating system). It looks like the pebble API provides some C functions (https://developer.getpebble.com/docs/c/group___standard_memory.html), I haven't had a big look, but it might be enough to use standard Rust.

Either way, zinc probably isn't needed as the pebble OS handles talking to the hardware.

photex, 2015-01-30:
Ah, ok well, that makes things a little easier for me!

0xc0170, 2015-01-30:
Agree with @bharrisau, use C API they provide. I had similar questions in the beginning, use the mailing list for questions or find us on IRC

If you find this as answered, please close it.

photex, 2015-01-30:
Meant to close this the last time. :) Big green buttons are difficult to avoid clicking!

closed 2015-01-30 by photex