hackndev/zinc #276: Build fails with errors

rnestler, 2015-02-17:
When I tried running
`rake PLATFORM=lpc17xx build_all`
I got some errors: https://gist.github.com/rnestler/ad705dd5542083a77d33#file-first
```
/home/roughl/proggen/projects/zinc/thirdparty/librlibc/src/lib.rs:26:1: 26:11 error: no_std is experimental
/home/roughl/proggen/projects/zinc/thirdparty/librlibc/src/lib.rs:26 #![no_std]
^~~~~~~~~~
/home/roughl/proggen/projects/zinc/thirdparty/librlibc/src/lib.rs:26:1: 26:11 help: add #![feature(no_std)] to the crate attributes to enable
/home/roughl/proggen/projects/zinc/thirdparty/librlibc/src/lib.rs:26 #![no_std]
^~~~~~~~~~ 
```
I could fix that one by using e2261b641d8634ed22688273cb0091c5cca785fa but I'm just getting more errors:
https://gist.github.com/rnestler/ad705dd5542083a77d33#file-second
```
/home/roughl/proggen/projects/zinc/src/ioreg/parser.rs:496:17: 496:22 error: type `syntax::parse::token::InternedString` does not implement any method in scope named `get`
/home/roughl/proggen/projects/zinc/src/ioreg/parser.rs:496           if !s.get().starts_with(prefix) {
                                                                           ^~~~~
```
Am I using a wrong version of rustc?
```
% rustc --version
rustc 1.0.0-nightly (b63cee4a1 2015-02-14 17:01:11 +0000)
```

bharrisau, 2015-02-18:
Does it build if you remove `.get()`? Looks like `.get()` was recently
replaced with a `Deref` impl.

awelkie, 2015-02-20:
What version of rustc is Zinc targeting? I haven't been able to get Zinc to build with rust nightlies for a few weeks now.

farcaller, 2015-02-20:
It is supposed to target nightly, but I think most of us got too frustrated
with rustc breaking every second day. I'm personally taking a timeout until
1.0, prototyping more of the platform tree in ruby now :-)

rnestler, 2015-02-21:
@bharrisau 
Removing all the `.get()` worked.
But now I get a
```error: Field target-pointer-width in target specification is required```

bharrisau, 2015-02-21:
Closing this. See #277

closed 2015-02-21 by bharrisau