hackndev/zinc #282: The attribute `phase` is currently unknown to the the compiler and may have meaning added to it in the future

ekiwi, 2015-03-18:
Hi,

zinc does not build with the latest rustc. The error message is the following:

~~~
The attribute `phase` is currently unknown to the the compiler and may have meaning added to it in the future
~~~

Check out the latest travis ci build from my fork which only has one commit that adds one white space to the README file in order to get travis to build zinc.

https://travis-ci.org/ekiwi/zinc/builds/54901512

I'm new to rust, so I have no idea what this error message is about, but I'm sure someone who knows the code will probably know what to do.

eKiwi

farcaller, 2015-03-18:
yup, this is known. The bigger issue so far is that `quote_method!()` is gone, breaking good part of ioreg, I'm working on it.

mcoffin, 2015-05-16:
FWIW, this is fixed in [mcoffin/new-build](/mcoffin/zinc/tree/new-build) and should work once my changes there get pulled (I'll open a PR when the build system is a little more complete).

mcoffin, 2015-06-01:
Fixed with the landing of #285

closed 2015-06-01 by mcoffin