hackndev/zinc #287: Document use of debugger when using Rust/Zinc

posborne, 2015-05-30:
I haven't tried it as of yet but I think there are ways for a GDB server to be set up for devices which exposes debug probes via JTAG/SWD (via, e.g., Segger J-Link) or CMSIS-DAP.  Since this is something that many embedded developers rely heavily on for regular development, I feel it would be useful to add a Wiki page or similar with pointers on how this can be made to work on a few different MCUs.

I can take a shot at figuring things out for the mbed LPC1768 which now supports the CMSIS-DAP protocol over USB.  I *think* that can be paired with OpenOCD to get a gdb server.  In theory, from there, you may be able to figure out a way to use gdb or even `rust-gdb`.

Even if the interface isn't "slick" out of the box, it can be quite nice to break and poke some registers.

farcaller, 2015-05-30:
I've tested lpc board with [pyOCD](https://github.com/mbedmicro/pyOCD) and gdb, and stm with openocd. In fact, good documentation about hw debugging is a nice idea, as I managed to spend unreasonable amount of time figuring it out. Specifically [this post](http://bgamari.github.io/posts/2014-08-23-swd-with-busblaster-and-openocd.html) of @bgamari was very useful.

posborne, 2015-05-31:
Thanks for the pointers @farcaller.  I got things kinda working under Linux with the latest release of OpenOCD, but it was extremely slow and buggy.  I had better luck with pyOCD (for CMSIS-DAP at least).  I have tested that from GDB and Eclipse.  The [EmbSysRegView](http://embsysregview.sourceforge.net/) plugin when debugging is very convenient as you can see all the peripheral registers for your board (the extension has a bunch of boards supported via CMSIS-SVD definitions, including the LPC175x6x).

Installation of pyOCD is not exactly streamlined, so I may see if I can improve that upstream (I have done lots of Python work).

The end result is pretty slick (I am somewhat partial to GUI debuggers).  I'll start by documenting how to set something like this up for the LPC1768.

[image]

0xc0170, 2015-05-31:
@posborne can you elaborate on that pyOCD? Or rather create an issue in pyOCd repo, if there's something we can improve.

I'll be happy to read that document once it's done!

posborne, 2015-05-31:
@0xc0170 No problem.  I am working on some changes now and will submit a Pull Request with a start to things when ready.  Here's what I think will improve things starting out:

1. Make setup.py declare requirements for each platform using `install_requires`
2. Add entry point for scripts (like GDB server)

With these two things in place, installation should be as easy as the following (on all platforms):

```
$ sudo pip install pyocd
$ pyocd-gdbserver
...
```

posborne, 2015-05-31:
I've started a page here: https://github.com/hackndev/zinc/wiki/Debugging-Zinc-Code-with-GDB-and-Eclipse.  It isn't complete yet and I'll continue to improve it later today.

farcaller, 2015-05-31:
> TODO: Building with debug symbols currently requires modifying Makefile

@posborne this is actually a bug, binaries are expected to be always built with those (as the bin/hex is stripped). Can you file a bug, or, better, a PR?

posborne, 2015-06-01:
@0xc0170, I've opened a PR against pyOCD that should greatly simplify installation and use of pyOCD for both library and executable (gdb server) users: https://github.com/mbedmicro/pyOCD/pull/149

posborne, 2015-06-02:
Initial documentation is now present, though limited to CMSIS-DAP debugging with pyOCD.  I'm closing this for now and we can add details later as people debug with new probes, etc.

closed 2015-06-02 by posborne