hackndev/zinc #303: Refactor zinc crate to support native build

farcaller, 2015-06-01:
As of now it's not possible to build zinc crate as it uses arm assembly. Still, zinc contains lots of code that would be interesting to unit test and validate coverage.

The proposed solution is to extract all machine-dependent code into new file hierarchy, namely `arch/`, and allow that to depend on feature flags.

Actual mcu code should be buildable for native arch as it's expected that it would be unit tested as well.

mcoffin, 2015-06-01:
Compiling the zinc crate natively works fine for me? `cargo test --lib --verbose`.

It won't work of you enable one of the platform features, as that will enable that ISR vector code for that platform (and therefore some assembly). Note that the stuff in `hal/platform` is not gated behind `#[cfg(feature = "platform")]` and can therefore be unit tested if you can find a way to get it to use dummy registers.

farcaller, 2015-06-01:
That's for test only, no? I mean, compile the dylib/rlib of zinc for native target.

farcaller, 2015-06-01:
Also, #302 will sort out the dummy registers part.

bharrisau, 2015-06-01:
I think the `#cfgs` for the asm code is currently just for tests. But
moving all of the unsafe ASM to a separate mod and documenting it (so we
don't get asm popping up everywhere) would be a good thing for hygiene.

mcoffin, 2015-06-01:
@bharrisau Regarding what you said in #305, i think a decent solution might be to come up with a system like rustc's "lang items". Then, i would agree, `#[cfg(...)]` to import the right impl (maybe within some `intrinsics` mod?)

farcaller, 2015-06-02:
> a system like rustc's "lang items"

would be nice, but the only way to implement something like it, as far as I know, is to have a stable path and `pub use` the `#[cfg]`-guarded version into it.

mcoffin, 2016-11-30:
Note for self and anyone reading: the `zinc` crate will now build for native, but that's just because all the mcu code sits behind cargo feature gates.