hackndev/zinc #306: queue code doesn't compile

awelkie, 2015-06-03:
After taking out the `#[cfg(cfg_multitasking)]` in `src/util/mod.rs`, `queue.rs` fails to compile.
```
➤ cargo test --lib
   Compiling zinc v0.1.0 (file:///home/allen/repos/zinc)
src/util/queue.rs:9:5: 9:21 error: unresolved import `core::ty::Unsafe`. Could not find `ty` in `core`
src/util/queue.rs:9 use core::ty::Unsafe;
                        ^~~~~~~~~~~~~~~~
src/util/queue.rs:12:5: 12:22 error: unresolved import `core::ptr::RawPtr`. There is no `RawPtr` in `core::ptr`
src/util/queue.rs:12 use core::ptr::RawPtr;
                         ^~~~~~~~~~~~~~~~~
src/util/queue.rs:15:5: 15:40 error: unresolved import `hal::cortex_m3::sched::NoInterrupts`. Could not find `sched` in `hal::cortex_m3`
src/util/queue.rs:15 use hal::cortex_m3::sched::NoInterrupts;
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
error: aborting due to 3 previous errors
Could not compile `zinc`.
```
It just looks like code rot Is there any reason to keep that `cfg` around?

mcoffin, 2015-06-03:
That `#[cfg(...)]` attribute should be moved to a cargo feature, but should be kept around if you don't want the scheduler.

The rest is code rot due to the instability of libcore. Much of the scheduler is pretty broken from what I gathered from #284.

farcaller, 2015-06-08:
Scheduler code needs a good refactoring that I did in a separate fork. Plan for it is to utilise POSIX signal() to test the userland scheduling code (which first needs to have zinc compilable for native).

mcoffin, 2015-06-08:
Ahh, so not really ready then as we're not yet sure how we're going to move all the architecture specific code out of the zinc crate. Gotcha.