hackndev/zinc #308: Add support for the Freescale KL25Z (KLxx)

posborne, 2015-06-03:
I have a few KL25Z FRDM boards laying around and think it would be an interesting board to add as it has a Cortex M0+ rather than an M3/M4.  The FRDM board from freescale can be bought for $12.95 and includes a few extra peripherals on board that are nice for testing.

Notes

* MCU: MKL25Z128VLK4
* On Board Peripherals:
  * MMA8451Q - 3-axis accelerometer (I2C)
  * PWM Controlled RGB LED
  * Capacitive touch sensor
* Easy programming via mbed standard; supports CMSIS-DAP out of the box for debugging

More info and links on the mbed page here: https://developer.mbed.org/handbook/mbed-FRDM-KL25Z.

I haven't dug into things enough to determine if there are levels of granularity/IP reuse beyond the following that might make sense:

* `src/hal/cortex_m0+`: Probably nothing unique starting out
* `src/hal/klxx': I need to look at the other kinetis processors more closely to determine if we can go more this generic.  This is the level of granularity used by mbed: https://github.com/mbedmicro/mbed/tree/master/libraries/mbed/targets/hal/TARGET_Freescale.

Here's the block diagram for KL2x processors from http://www.freescale.com/webapp/sps/site/prod_summary.jsp?code=KL2x:

[image]

Thanks for any input.  With this development, I would also like to see how effectively we can leverage the work for `ioregs!` generation form the CMSIS-SVD for the chip family (assuming I can locate it) based on https://github.com/hackndev/zinc/issues/259 and https://github.com/hackndev/zinc/pull/267.

0xc0170, 2015-06-03:
Yes, you can go generic as in mbed those KLxx, although they differ at least in spi as I recall, and there are newer KLxx devices which contain a bit different serial, spi and I recall there's at least one "new" peripheral for serial communication.

I was planning a while ago to add KL25Z, I might help you, to get back to rust :smile:

bharrisau, 2015-06-03:
You have a couple of levels to thinl about. The peripherals are usually 90%
the same, some optional features and different capacities result in
additional registers. I'm still not sure the best way to deal with these
minor differences in peripherals.

Then you also have the system controller that controls the clock for each
peripheral. This changes depending on which peripherals are around. This
would be best as a compile time thing, as more than half of its
configuration is only known at compile time when the device is specified.