hackndev/zinc #327: RFC: generate ioregs! macros from SVD files

niclashoyer, 2015-07-04:
After discussion [here](https://github.com/hackndev/zinc/pull/325#discussion_r33886436) it seems that we could save a lot of work if we generate `ioregs!` macros from [SVD files](https://www.keil.com/pack/doc/CMSIS/SVD/html/svd__outline_pg.html).

Based on the great work of @posborne I threw together a [python script](https://gist.github.com/niclashoyer/06e9c26076af1b2fd485), that generates register definitions for STM32F4. It does not handle more complex cases with enumerated values, though.

One caveat is, that one has to be careful regarding the licenses of the SVD files and check if they are compatible with the Apache license or at least allow derivates to be Apache licensed.

Any ideas on how to integrate such a python generator in the zinc build system?

niclashoyer, 2015-07-04:
related: #326

farcaller, 2015-07-04:
there's little sense to not commit the generated files in tree, really (like I did in #318).

niclashoyer, 2015-07-04:
ah, so we already got the ruby implementation of the svd parser/generator. I'll take a look at it in `support/`.

posborne, 2015-07-04:
This is also in-progress: https://github.com/posborne/zinc/tree/svd-generator-python.  Haven't been able to find much time lately to work on it more.

farcaller, 2015-07-09:
Recapping the discussion we had here and in 326, I guess this is how we should proceed.

* we don't modify SVD files to store them in zinc tree
* every platform gets a dedicated `regs.rs` with all the ioregs, generated either manually, or from svd (whatever that is, a few people still need to cross-check it with docs)
* ioreg macro becomes our source of truth for a platform, we can always compile it into whatever descriptive format we want if need
* it is fine to modify those `regs.rs` files to get better readability and constant naming