hackndev/zinc #328: How can I use ioreg from Cargo?

russellmcc, 2015-07-06:
I don't know that cargo is really set up for multi-crate repos like this.  Why aren't independently useful things like ioreg split up into separate repositories so they can be accessed from cargo?

farcaller, 2015-07-06:
The last time I checked there were issues with uploading crates that are used against nightly to crates.io. Is ti still the case? I'm all in to publish ioregs and volatile_cell.

mcoffin, 2015-07-07:
```toml
[dependencies.ioreg]
git = "https://github.com/hackndev/zinc.git"
path = "ioreg"
```

That will get you ioreg until we beleive it's stable enough to bother publishing to crates.io.

mcoffin, 2016-11-30:
Closed as the question has been answered.

closed 2016-11-30 by mcoffin