hackndev/zinc #331: Modify examples to be dedicated crates

farcaller, 2015-07-14:
As a followup to #330, we need to refactor all the examples to be dedicated crates. This also shows how hard zinc is to use for external users. I'd expect a zinc app to be just one more crate. It is unreasonable to expect the users to download zinc source and add a new "example" entry.

Here's a [sample commit](https://github.com/hackndev/zinc/commit/7b876c112c279930e77d730e1040258e0c315919) that is broken so far:

```
error: linking with `arm-none-eabi-gcc` failed: exit code: 1
note: layout.ld:5: undefined symbol `isr_nmi' referenced in expression
```

mcoffin, 2015-07-14:
Does posborne/zinc-example-lpc17xx not work anymore?

farcaller, 2015-07-14:
they do. The bundled examples are broken though.

mcoffin, 2015-07-14:
> It is unreasonable to expect the users to download zinc source and add a new "example" entry.

Ah I was just confused by this sentence.

farcaller, 2015-07-14:
Right. I forgot about that wiki page when I was writing it.

farcaller, 2015-07-18:
Landed all in #318.

closed 2015-07-18 by farcaller