hackndev/zinc #332: Could not compile `ioreg`.----build failed on macos

enginespot, 2015-07-17:
```
export PLATFORM="lpc17xx"                        
➜  zinc git:(master) ./support/build-jenkins.sh
 * rustc version: rustc 1.3.0-nightly (e4e93196e 2015-07-14)
checking for arm-none-eabi-gcc... arm-none-eabi-gcc
checking for arm-none-eabi-strip... arm-none-eabi-strip
checking for arm-none-eabi-objcopy... arm-none-eabi-objcopy
checking for arm-none-eabi-objdump... arm-none-eabi-objdump
checking for arm-none-eabi-ar... arm-none-eabi-ar
configure: creating ./config.status
config.status: creating Makefile
config.status: creating .cargo/config
   Compiling core v1.0.0 (https://github.com/hackndev/rust-libcore#d9dc4f94)
     Running `rustc /Users/edwin/.cargo/git/checkouts/rust-libcore-a3e1d25b35d5cd12/master/rust/src/libcore/lib.rs --crate-name core --crate-type lib -g -C metadata=59996a419f60f2f9 -C extra-filename=-59996a419f60f2f9 --out-dir /Users/edwin/Projects/zinc/target/thumbv7m-none-eabi/debug/deps --emit=dep-info,link --target thumbv7m-none-eabi -C ar=arm-none-eabi-ar -C linker=arm-none-eabi-gcc -L dependency=/Users/edwin/Projects/zinc/target/thumbv7m-none-eabi/debug/deps -L dependency=/Users/edwin/Projects/zinc/target/thumbv7m-none-eabi/debug/deps -Awarnings`
   Compiling ioreg v0.1.0 (file:///Users/edwin/Projects/zinc)
     Running `rustc ioreg/src/lib.rs --crate-name ioreg --crate-type dylib -C prefer-dynamic -g -C metadata=11ccf67161f638aa -C extra-filename=-11ccf67161f638aa --out-dir /Users/edwin/Projects/zinc/target/debug/deps --emit=dep-info,link -L dependency=/Users/edwin/Projects/zinc/target/debug/deps -L dependency=/Users/edwin/Projects/zinc/target/debug/deps --extern syntaxext_lint=/Users/edwin/Projects/zinc/target/debug/deps/libsyntaxext_lint-1bf554c3b16db5a8.dylib`
   Compiling zinc v0.1.0 (file:///Users/edwin/Projects/zinc)
     Running `/Users/edwin/Projects/zinc/target/debug/build/zinc-3e19c03f5adab03d/build-script-build`
       Fresh platformtree v0.1.0 (file:///Users/edwin/Projects/zinc)
       Fresh syntaxext_lint v0.1.0
       Fresh macro_zinc v0.1.0 (file:///Users/edwin/Projects/zinc)
       Fresh macro_platformtree v0.1.0 (file:///Users/edwin/Projects/zinc)
error: internal compiler error: unexpected panic
note: the compiler unexpectedly panicked. this is a bug.
note: we would appreciate a bug report: https://github.com/rust-lang/rust/blob/master/CONTRIBUTING.md#bug-reports
thread 'rustc' panicked at 'index out of bounds: the len is 60 but the index is 1081', ../src/libcollections/vec.rs:1376

stack backtrace:
   1:        0x104cb9785 - sys::backtrace::write::ha27d712b848aedc4Eos
   2:        0x104cc2623 - panicking::on_panic::h9f9b299259faeda5FMw
   3:        0x104c7d7d2 - rt::unwind::begin_unwind_inner::h34588507fea4f430Ruw
   4:        0x104c7e529 - rt::unwind::begin_unwind_fmt::haf01b18ac46172faXtw
   5:        0x104cc21ac - rust_begin_unwind
   6:        0x104d180c5 - panicking::panic_fmt::hc126a8e23373fbe7neC
   7:        0x104d12885 - panicking::panic_bounds_check::hb60c81cc5e896646tdC
   8:        0x109eb89b7 - util::interner::StrInterner::get::h0d156d7255eec8e6Fpa
   9:        0x105399874 - parse::token::get_name::hf3a8b6aa848c01f6YhT
  10:        0x1053997a8 - parse::token::get_ident::h64f49d6a7635d4a6fiT
  11:        0x105399647 - Pass.LintPass::check_expr::h0bd92942e09e09991aa
  12:        0x101e882e4 - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_expr::hf7f8aca8e25f497cKOu
  13:        0x101e885fb - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_expr::hf7f8aca8e25f497cKOu
  14:        0x101e885fb - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_expr::hf7f8aca8e25f497cKOu
  15:        0x101e885fb - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_expr::hf7f8aca8e25f497cKOu
  16:        0x101e885fb - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_expr::hf7f8aca8e25f497cKOu
  17:        0x101e885fb - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_expr::hf7f8aca8e25f497cKOu
  18:        0x101e8c3c3 - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_arm::h99d850818547dc0092u
  19:        0x101e88758 - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_expr::hf7f8aca8e25f497cKOu
  20:        0x101e88c0e - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_fn::hd556c5b80c885a6fQQu
  21:        0x101e93df8 - visit::walk_impl_item::h294280462252581110
  22:        0x101e93689 - lint::context::Context<'a, 'tcx>::with_lint_attrs::h2160803342945724918
  23:        0x101e840ad - lint::context::Context<'a, 'tcx>::with_lint_attrs::h3132791760906412596
  24:        0x101e88e9d - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_mod::h3ca23211f356a12bSZu
  25:        0x101e83ea5 - lint::context::Context<'a, 'tcx>::with_lint_attrs::h3132791760906412596
  26:        0x101e88e9d - lint::context::Context<'a, 'tcx>.Visitor<'v>::visit_mod::h3ca23211f356a12bSZu
  27:        0x101e9668f - lint::context::check_crate::hc0e4f890e229305cRjv
  28:        0x1011b9003 - driver::phase_3_run_analysis_passes::closure.15896
  29:        0x1011b3882 - middle::ty::ctxt<'tcx>::create_and_enter::h13599737314050492207
  30:        0x1011aea1d - driver::phase_3_run_analysis_passes::h13797044866949165922
  31:        0x10118e1c7 - driver::compile_input::h407a10a539c7270cTba
  32:        0x101273d9f - run_compiler::hed8febc643a5bd02A7b
  33:        0x101271653 - boxed::F.FnBox<A>::call_box::h5082374284048568258
  34:        0x101270e17 - rt::unwind::try::try_fn::h4270298375276078235
  35:        0x104d50318 - rust_try_inner
  36:        0x104d50305 - rust_try
  37:        0x104cabcf5 - rt::unwind::try::inner_try::h9dc445b523024767Kqw
  38:        0x101271048 - boxed::F.FnBox<A>::call_box::h14263195620542072551
  39:        0x104cc10ad - sys::thread::Thread::new::thread_start::h235524d294e76343UOv
  40:     0x7fff97c20267 - _pthread_body
  41:     0x7fff97c201e4 - _pthread_start

Build failed, waiting for other jobs to finish...
Could not compile `ioreg`.

Caused by:
  Process didn't exit successfully: `rustc ioreg/src/lib.rs --crate-name ioreg --crate-type dylib -C prefer-dynamic -g -C metadata=11ccf67161f638aa -C extra-filename=-11ccf67161f638aa --out-dir /Users/edwin/Projects/zinc/target/debug/deps --emit=dep-info,link -L dependency=/Users/edwin/Projects/zinc/target/debug/deps -L dependency=/Users/edwin/Projects/zinc/target/debug/deps --extern syntaxext_lint=/Users/edwin/Projects/zinc/target/debug/deps/libsyntaxext_lint-1bf554c3b16db5a8.dylib` (exit code: 101)
```

farcaller, 2015-07-17:
Can't reproduce so far on same compiler and clean HEAD, will look into it over the weekend.

farcaller, 2015-07-19:
Another stack trace after `#![plugin(syntaxext_lint)]` was removed:

https://gist.github.com/enginespot/93e288023bfc04bcf58e

farcaller, 2016-01-06:
Looks fine now.

closed 2016-01-06 by farcaller