hackndev/zinc #333: Fix makefile to build cargoized examples

farcaller, 2015-07-18:
Makefile is currently broken from #318 and will not build examples as expected. The whole its existence is slightly questionable now, as it's basically pre and post-processing around cargo. Maybe we need to make a simple wrapper around cargo anyway (sounds like a reasonable option given how cargo isn't that much cross-build friendly)?

mcoffin, 2015-08-17:
I agree, especially since a lot of that processing is platform-specific. Personally, however, I find the listing files *incredible* for debugging, so I would want a version kept around anyways.

The other thing it does is hide our slightly-not-idiomatic "feature" usage in cargo. I think it would be worth making an RFC for refactoring how we go about separating out all the various platforms and stuff (which really should be done alongside a better, more well-thought-out implementation of #286). It might even be prudent to actually set a time and get together at some point on IRC and talk though it. I've banged my head at coming up with a completely-idiomatic adaptable solution and it's actually a relatively hard problem to solve while still adhering to the 0-overhead philosophies of Rust.

farcaller, 2015-12-14:
docs updated to use proper cargo commands in #351

closed 2015-12-14 by farcaller