hackndev/zinc #334: Zinc fails to build due to the removal of get_ident()

LamentOfHeroes, 2015-07-31:
It seems that rust-lang/rust@00a5e66f818ad9d79cc4425f5564c7b07e3213a6 removed get_ident(), preventing Zinc from compiling. It was used in these files:
ioreg/src/builder/{accessors.rs,getter.rs,register.rs,setter.rs,union.rs,utils.rs}
ioreg/src/parser.rs
macro_platformtree/src/lib.rs
platformtree/src/parser.rs

The rust version I used when I discovered this was rustc 1.3.0-nightly (4dfe7a16c 2015-07-30)

LamentOfHeroes, 2015-07-31:
Unsure if it helps, but I've fixed what I can in pull request #335.

mcoffin, 2015-07-31:
Ok the upstream bug is fixed now. It was just the same problem effecting one of our dependent syntax extension crates. I'm going to tell travis to rebuild your pr.

commit b55adef3 (2015-07-31): Merge pull request #335 from LamentOfHeroes/master

Fixes #334
closed 2015-07-31 by mcoffin