hackndev/zinc #341: Consider utilizing yasha for ioreg generation from SVD

kblomqvist, 2015-09-08:
I have been developing code generator with SVD to ioreg case in mind. I know that you already have some tooling to generate ioreg macros from SVD, but I though that you would still be interested in to see what I have made.

The example project can be found here: https://github.com/kblomqvist/nrf51.rs

- `nrf51.rs.jinja` is a template file for the generated `nrf51.rs`
- `nrf51.jinja-ext` contains the SVD parser and few template filters. The core of the SvdFile parser is placed in `cmsis.py`.
- From template you can also easily enable/disable whether you want to have register fields and/or enumerated values generated, or descriptions printed.

The parser also supports cluster elements, which aren't addressed by the current tooling.

kblomqvist, 2015-10-03:
FYI, I have made several improvements to the example project since filing the issue.

kblomqvist, 2015-12-27:
Update:

The mentioned CMSIS-SVD parser is now part of the default parsers of Yasha, which is a code generator based on Jinja2 template engine.

Lately I have been working with the SCons integration of Yasha for C and came up with the custom builder. Examples can be found here, https://github.com/kblomqvist/yasha#sconstruct-scons. Shortly it's possible to generate C files (also headers) so that SCons can resolve the dependencies coming from the generated code.

However, what I have understood is that Cargo doesn't support *"custom dependency builders"* so it's not possible to do similar integration for Rust. Or is it?

farcaller, 2015-12-27:
Cargo supports custom build steps to some extent, but I don't think that regenerating all SVDs should be part of it really.

kblomqvist, 2015-12-27:
I was thinking that only the target SVD would be generated during the first build and regenerated after the clean. I'm assuming that the build system can know the target MCU.