hackndev/zinc #347: blink_k20 fails to build

plasticchris, 2015-10-19:
The read me at the root indicates to use the makefile to build; this doesn't work:

    chris:zinc chrisjohnson$ EXAMPLE_NAME=blink_k20 make build
    cd .
    cargo build --example blink_k20 --release --target=thumbv7em-none-eabi --verbose --features mcu_k20
     Downloading num v0.1.27
     Downloading expectest v0.3.1
    no example target named `blink_k20`
    make: *** [target/thumbv7em-none-eabi/release/examples/blink_k20] Error 101

Further reading indicated I needed to 'care build' in the example directory, which initially failed due to my rust environment being stable, tried beta with the same error, so I tried nightly and got this new error:

    chris:blink_k20 chrisjohnson$ cargo build
       Compiling ioreg v0.1.0 (file:///Users/chrisjohnson/zinc/examples/blink_k20)
       Compiling core v1.0.0 (https://github.com/hackndev/rust-libcore#d9dc4f94)
    /Users/chrisjohnson/zinc/ioreg/src/builder/union.rs:179:22: 179:36 error: `ast::StructDef` does not name a structure [E0422]
    /Users/chrisjohnson/zinc/ioreg/src/builder/union.rs:179     let struct_def = ast::StructDef {
                                                                                 ^~~~~~~~~~~~~~
    /Users/chrisjohnson/zinc/ioreg/src/builder/union.rs:179:22: 179:36 help: run `rustc --explain E0422` to see a detailed explanation
    /Users/chrisjohnson/zinc/ioreg/src/builder/register.rs:166:13: 166:34 error: unresolved name `ast::TupleVariantKind` [E0425]
    /Users/chrisjohnson/zinc/ioreg/src/builder/register.rs:166       kind: ast::TupleVariantKind(Vec::new()),
                                                                           ^~~~~~~~~~~~~~~~~~~~~
    /Users/chrisjohnson/zinc/ioreg/src/builder/register.rs:166:13: 166:34 help: run `rustc --explain E0425` to see a detailed explanation
    error: aborting due to 2 previous errors

It looks like some issue with ioreg?

farcaller, 2015-10-19:
Structs broke down in fun and unexpected way apparently, opened a PR to track a fix.

farcaller, 2015-12-14:
fixed in #351

closed 2015-12-14 by farcaller