hackndev/zinc #349: Port to NXP LPC2388

aleksander, 2015-12-01:
Hello.
I have SK-MLPC2388 evaluation board which is based on NXP LPC2388 chip (ARM7TDMI-S, 512k flash, 64k RAM).
I want to port Zinc to it.
Is it possible?
How much effort I have to make to run 'hello world'?
Are there any tutorials or other docs about porting Zinc?
I would be grateful for any help.

farcaller, 2015-12-02:
It's clearly possible but I don't think it's worth adding another architecture to zinc right now. Ioreg framework became a bit of a mess and PT is plainly broken.

I'm prototyping a new approach that would hopefully provide a more stable API to define hardware registers API, expect some updates before EOY.

closed 2015-12-02 by farcaller

mcoffin, 2015-12-02:
@farcaller Good to hear. I've been playing around a little too but have been too busy with work to really finish anything.