hackndev/zinc #358: Consider switching to official rust style (rustfmt zinc)

posborne, 2015-12-16:
Putting out a feeler.  I think using the standard style is probably a good idea and makes contributing easier (in addition to making my emacs configuration easier).

https://github.com/rust-lang/rust/tree/master/src/doc/style

rustfmt can probably do a majority of the work, although it still needs some handholding in my experience to prevent it from going in the weeds in edge cases.  With `cargo install` we can be `rustfmt`ing in no time.

I don't think there are too many major outstanding PRs that would need to be reworked, so now seems as good a time as any to make the switch if this suggestion is amenable.

mcoffin, 2015-12-16:
For reference, [previous discussion](https://github.com/hackndev/zinc/pull/285#issuecomment-105349159).

**TL;DR**: This would :hocho: `git-blame`

posborne, 2015-12-16:
@mcoffin.  You're right of course on `git blame` but I think it probably is worth it.  `git blame` does have options for ignoring whitespace changes (`-w`) which might be enough to make the reformat less of a big deal (given how often blames are actually performed in practice in my experience).  I'd like to think that Zinc has a long, bright future in front of it and reformatting may end up being one of the smaller changes in the grand scheme of things.

I think (and hope?) the cognitive dissonance of a unique style is likely to only increase over time as the standard style becomes more and more standard (e.g. I bet IDEs will not even make much of the style for Rust code configurable).

mcoffin, 2015-12-16:
Don't get me wrong, I'm all for it. Gonna be some work for somebody though.

And yea... we need to have an RFC party or something...

bharrisau, 2015-12-16:
This might be best tackling when splitting things into new repos. I think
@farcaller mentioned this in the last 24hrs.

Have we removed all the `#ignore doc_comments` yet too?

farcaller, 2015-12-16:
I'm all in for it, but is rustfmt stable enough now?

posborne, 2015-12-16:
@farcaller It's pretty good but you still need to look at the diffs to keep things in check.  I ran it on a few of my projects the other day with pretty good results.

mcoffin, 2015-12-16:
> you still need to look at the diffs to keep things in check

*touches nose*

farcaller, 2015-12-16:
I want to kill `ioreg!` though... :fire:

mcoffin, 2015-12-16:
> I want to kill `ioreg!` though... :fire:

Like I said earlier and on IRC, we need to have an RFC party and make some tech decisions so that we can start working towards common goals.

farcaller, 2015-12-16:
Makes sense. Let me fire up a dedicated ticket for that.