hackndev/zinc #359: Zinc IRC RFC party

farcaller, 2015-12-16:
Pinging in @bgamari @0xc0170 @bharrisau @kvark @posborne @mcoffin and @errordeveloper. Are you in for a few hours of discussion on project goals?

bharrisau, 2015-12-16:
Sure, GMT+11 here. Have some dot points of what to discuss?

kvark, 2015-12-16:
Sorry, I'll pass on the discussion, will follow the thread.

mcoffin, 2015-12-16:
I'm in. GMT-7.

posborne, 2015-12-16:
I'm GMT-6 but pretty flexible as well.

bharrisau, 2015-12-16:
Gitter is an option too if IRC is too onerous for some.

lizardo, 2015-12-17:
I just started playing with zinc a few weeks ago (and with Rust itself for a few months), but I would like to join to get a better understanding of the future steps for the project.

GMT-4 here. I prefer night time, but just say the day/time/place and I will try to join.

Either gitter or IRC is fine (is there an official IRC channel?). It would be nice if someone could take chat logs for those who can't join.

posborne, 2015-12-17:
@lizardo #rust on moznet (irc.mozilla.org)

0xc0170, 2015-12-17:
For me GMT. Seems like we are quite scattered around the world

> It would be nice if someone could take chat logs for those who can't join.

+1, I assume they will result in RFC or tracking issues. we will be able to catch up

pablochacin, 2015-12-17:
yes, share log, please.

---------------------------
Pablo Chacin
CTO
SenseFields SL
Tlf (+34) 93 250 45 98
Gran Via 674, principal 1º
08010 Barcelona, Spain
http://www.sensefields.com

En compliment del que disposa la Llei Orgànica de Protecció de Dades
15/1999 i el seu reglament, Sensefields, S.L. us informa que les vostres
dades personals seran tractades i incorporades als nostres sistemes
informàtics i documentals, dels quals és titular  aquesta empresa. Si voleu
podeu exercir els drets d'accés, rectificació, cancel·lació i oposició
previstos a la llei, adreçant un escrit amb la fotocòpia del DNI a
Sensefields, S.L. Gran Via Corts Catalanes, 674 Principal 1ª - 08010
Barcelona (Barcelona) o bé per e.mail a info@sensefields.com

Aquest missatge va dirigit, de manera exclusiva, al seu destinatari, i
conté informació confidencial i privilegiada. En cas de rebre aquest
missatge per error, prego que ens ho comuniquin de forma immediata
mitjançant resposta per correu electrònic, o a través del telèfon 0034 93
250 45 98, i procedeixi a la seva eliminació.

En cumplimiento de lo dispuesto en la Ley Orgánica de Protección de Datos
15/1999 y su reglamento, Sensefields, S.L.  le informa que sus datos
personales serán tratados e incorporados a nuestros sistemas informáticos y
documentales, de los que es titular esta empresa. Si desea puede ejercer
los derechos de acceso, rectificación, cancelación  y oposición previstos
en la ley, dirigiendo un escrito con la fotocopia del DNI a Sensefields,
S.L. Gran Via Corts Catalanes, 674 Principal 1ª - 08010 Barcelona
(Barcelona) o bien por e.mail a info@sensefields.com

Este mensaje va dirigido, de manera exclusiva, a su destinatario y contiene
información confidencial y privilegiada. En caso de recibir este mensaje
por error, ruego que nos lo comuniquen de forma inmediata mediant respuesta
por correo electrónico, o a través del teléfono 0034 93 250 45 98, y
proceda a su eliminación.

In compliance with The Law of Data Protection Act 15/1999 and its
regulations, Sensefields, S.L.  informs you that your personal data will be
processed and stored in our computer systems and documentaries owned by
this company. If you can exercise your rights of access, rectification,
cancellation and opposition under the Act, by writing the photocopy of ID
to Sensefields, S.L. Gran Via Corts Catalanes 674 Pral 1ª 08010 -
(Barcelona) (Barcelona) or by email to info@sensefields.com

This message was directed exclusively at the recipient and contains
privileged and confidential information. If you receive this message in
error, I beg to inform us immediately by reply email or by phone 0034 93
250 45 98, and proceed to their elimination.

mcoffin, 2015-12-17:
Adding this to the discussion:
* [Generalize platformtree](https://github.com/hackndev/zinc/issues/286) and less-painful generalization of MCUs. 

Current thoughts on this: It might be possible to generalize the mcu like this with Rust's type system (using marker traits + empty enums) since it's turing complete instead of relying on compile-time macros, feature flags, and `#[cfg(..)]`-gating.

farcaller, 2016-01-06:
Filed #365 with more details on suggested move from `ioregs!`