hackndev/zinc #365: Pre-compiled ioregs RFC

farcaller, 2016-01-06:
This is one of the topics I want to discuss as part of RFC party (#359), and I finally cleaned up code enough to share it.

https://github.com/farcaller/zinc-hal-lpc11xx ([api doc](http://farcaller.github.io/zinc-hal-lpc11xx/zinc_hal_lpc11xx/ioregs/index.html)) provides a Cargoized hal layer for single MCU family (actually only ioregs now). It's generated with yasha (#341) with some modifications, most notably additional renaming/cleanup step (you can see the modification rules [here](https://github.com/farcaller/zinc-hal-lpc11xx/blob/master/names.yml)). It also fixes #344 and makes few modifications to the layout, most notably it supports sparse peripherals by addressing not the peripheral but individual registers.

bharrisau, 2016-01-06:
Significantly less magic (which is a good thing). Having docs are good, but it would be good if the docs were more modular. That is, the front page only has the high level structs - ADC, CT16B0, CT32B0, etc.

Then under ADC you have ADC_cr, ADC_dr0, ADC_dr1, etc. And the *_Get and *_Update are hidden another layer down.

I'd suggest we also document a guideline on how much of the module semantics is included, otherwise I can forsee us ending up with a [poor] copy of the device reference manual in the module docs.

Am I correct in assuming the HAL layer lives here too, and the Zinc crate will define some general traits that `zinc-hal-lpc11xx` would implement?

farcaller, 2016-01-06:
> Then under ADC you have ADC_cr, ADC_dr0, ADC_dr1, etc. And the *_Get and *_Update are hidden another layer down

good idea

> I'd suggest we also document a guideline on how much of the module semantics is included

well, that's what we have in SVD, should be a very basic description, right?

> Am I correct in assuming the HAL layer lives here too

Yep, I'll clean that part in a few days. I got problems making this work with crates.io version of volatile_cell.

farcaller, 2016-01-16:
Things are progressing slower than I'd want, but anyways, here's the deal:

* https://github.com/farcaller/pioreg is the set of templates to generate ioregs
* https://github.com/kblomqvist/yasha/pull/5 is a patch to yasha to make that work

kblomqvist, 2016-01-16:
I'm ok :)

kblomqvist, 2016-01-17:
I made a new pip release of yasha, so the needed changes are more readily available.
https://github.com/kblomqvist/yasha/releases/tag/1.5