hackndev/zinc #370: Compilation fail on current rustc nightly (1.9.0)

Gyscos, 2016-03-11:
Libsyntax at least has changed, and many compilation errors result from that.

Some changes are small (things like `syntax::ast::PatLit` being renamed to `syntax::ast::PatKind::Lit`, others are slightly larger (like some `syntax::ast::ItemKind::Impl.5` moving from `Vec<P<ImplItem>>` to `Vec<ImplItem>`).

thenewwazoo, 2016-03-12:
Additionally, [this commit](https://github.com/oli-obk/rust/commit/bfa66bb389ce1c7ce4aff09d1842b3428015bd4d) removes sign from the AST, which I'm not sure how to handle in ioreg/src/builder/utils.rs:30 (and more places besides).

thenewwazoo, 2016-03-12:
It's a totally naive suggestion, but could something like [syntex](https://github.com/serde-rs/syntex) be used instead of libsyntax?

thenewwazoo, 2016-03-12:
I've made an [exceedingly naive](https://github.com/thenewwazoo/zinc/tree/update-for-rust-1.9) attempt at resolving all the issues I can manage.

Gyscos, 2016-03-13:
Yeah, those come from the removal of the `P` indirection.

richo, 2016-03-16:
I created a PR that fixes many of these bugs. I'm currently working on where enum variants are specified to be u32 (I presume this is because they're `repr`d to the width of the registers) but sadly many primitive ops are defined on `usize`

Gyscos, 2016-04-07:
After updating to the new `libsyntax::ast` enums, compiling examples still give the following two kinds of errors (many errors of each kind):

```
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:172:3: 291:6 error: attribute should be applied to function [E0518]
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:172   ioregs!(PWM1@0x40018000 = {
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:173     /// Interrupt Register. The IR can be written to clear
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:174     /// interrupts. The IR can be read to identify which of eight
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:175     /// possible interrupt sources are pending.
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:176     0x00 => reg32 ir {
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:177       0  => irq_mr0,  //= Interrupt flag for PWM match channel 0.
```

```
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:193:10: 193:11 error: mismatched types:
 expected `u32`,
    found `usize` [E0308]
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:193          0 => DISABLED,
                                                              ^
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:172:3: 291:6 note: in this expansion of ioregs! (defined in /home/gyscos/fetched/zinc/src/lib.rs)
/home/gyscos/fetched/zinc/src/hal/lpc17xx/pwm.rs:193:10: 193:11 help: run `rustc --explain E0308` to see a detailed explanation
```

gyscos, 2016-09-24:
Zinc now (mostly) compiles with a modern compiler.

closed 2016-09-24 by gyscos