hackndev/zinc #375: Default build target for certain examples

Grawp, 2016-05-23:
The MCU specific examples have default features set in their Cargo.toml like
`default = ["mcu_stm32f4"]`
I propose also directly adding symlink to JSON target description and `.cargo/config` with build target like
`[build]
target = "thumbv7em-none-eabi"`

Such example could be then build just by running cargo build --release.
The generic examples like _empty_ would retain status quo (i.e. no default build target as well as no default features set),

I can make a pull request if somebody is interested in this idea.

farcaller, 2016-05-23:
Makes perfect sense, I'd appreciate a PR.

mcoffin, 2016-11-29:
This appears to have happened in d097403961421fe1d2062e66f1878d78c523bb99, which landed in #376, so I'm going to close this. Thanks!

closed 2016-11-29 by mcoffin