hackndev/zinc #379: Build of blink-stm32f4 fails with target specification issues.

mwm, 2016-06-23:
Ok, environment:

```
$ pwd
/export/mwm/src/github/zinc/examples/blink_stm32f4
$ which rustc
/export/mwm/.cargo/bin/rustc
$ rustc --version
rustc 1.10.0-nightly (8da2bcac5 2016-04-28)
$ file thumbv7em-none-eabi.json 
thumbv7em-none-eabi.json: symbolic link to ../../thumbv7em-none-eabi.json
$ 
```

But when I try and build this example:

```
bhuda(3)% cargo build --verbose --target=thumbv7em-none-eabi --features mcu_stm32f4 --release
error: failed to run `rustc` to learn about target-specific information

Caused by:
  Process didn't exit successfully: `rustc - --crate-name _ --crate-type dylib --crate-type staticlib --crate-type bin --print=file-names --target thumbv7em-none-eabi` (exit code: 101)
--- stderr
error: Error loading target specification: Could not find specification for target "thumbv7em-none-eabi"
```

At a guess, the problem is that something in the nightly channel changed, but I'm not sure. Maybe trying with a version that works for someone else would be useful. Could someone give me a version hash?

And of course, is there any other information you need from me in order to resolve this?

dhylands, 2016-09-05:
I've been able to build the blink_stm32f4 example by using rustup to override my toolchain to nightly-2016-05-24

mcoffin, 2016-11-29:
This appears to build for me as of e68f861. I'm going to close this. If you're still having an issue, feel free to re-open and I'll take a closer look.

closed 2016-11-29 by mcoffin