hackndev/zinc #380: ioreg broke with rust nightly-2016-06-10+

archaelus, 2016-06-24:
Hi all, something changed in rust nightly between 2016-06-09 and 2016-06-10 when building zinc:

`rustup run nightly-2016-06-09 cargo build --features mcu_k20` works fine:
```
rustup run nightly-2016-06-09 cargo build --verbose --features mcu_k20
   Compiling rlibc v0.1.4 (https://github.com/hackndev/rlibc?branch=zinc#620489f8)
     Running `rustc Home/.cargo/git/checkouts/rlibc-22ab3e9e80121e06/zinc/src/lib.rs --crate-name rlibc --crate-type lib -g -C metadata=961f6a1dbd8e6b29 -C extra-filename=-961f6a1dbd8e6b29 --out-dir Home/projects/lrn2rst/zinc/target/debug/deps --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps --cap-lints allow`
   Compiling volatile_cell v1.0.0 (file://Home/projects/lrn2rst/zinc/volatile_cell)
     Running `rustc volatile_cell/lib.rs --crate-name volatile_cell --crate-type rlib -g -C metadata=6eaf9a8b4cfc05df -C extra-filename=-6eaf9a8b4cfc05df --out-dir Home/projects/lrn2rst/zinc/target/debug/deps --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps`
   Compiling ioreg v0.1.0 (file://Home/projects/lrn2rst/zinc/ioreg)
     Running `rustc ioreg/src/lib.rs --crate-name ioreg --crate-type dylib -C prefer-dynamic -g -C metadata=4302d6378f0cb21c -C extra-filename=-4302d6378f0cb21c --out-dir Home/projects/lrn2rst/zinc/target/debug/deps --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps`
   Compiling zinc v0.1.0 (file://Home/projects/lrn2rst/zinc)
     Running `rustc build.rs --crate-name build_script_build --crate-type bin -g --cfg feature=\"cpu_cortex-m4\" --cfg feature=\"mcu_k20\" --out-dir Home/projects/lrn2rst/zinc/target/debug/build/zinc-d3eb462b2a6ccfcc --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps`
     Running `Home/projects/lrn2rst/zinc/target/debug/build/zinc-d3eb462b2a6ccfcc/build-script-build`
     Running `rustc src/lib.rs --crate-name zinc --crate-type lib -g --cfg feature=\"cpu_cortex-m4\" --cfg feature=\"mcu_k20\" --out-dir Home/projects/lrn2rst/zinc/target/debug --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps --extern rlibc=Home/projects/lrn2rst/zinc/target/debug/deps/librlibc-961f6a1dbd8e6b29.rlib --extern ioreg=Home/projects/lrn2rst/zinc/target/debug/deps/libioreg-4302d6378f0cb21c.dylib --extern volatile_cell=Home/projects/lrn2rst/zinc/target/debug/deps/libvolatile_cell-6eaf9a8b4cfc05df.rlib -L native=Home/projects/lrn2rst/zinc/target/debug/build/zinc-d3eb462b2a6ccfcc/out`
src/lib.rs:17:45: 17:57 warning: unused or unknown feature, #[warn(unused_features)] on by default
src/lib.rs:17 #![feature(core_intrinsics, core_slice_ext, core_str_ext)]
                                                          ^~~~~~~~~~~~
src/drivers/lcd/c12332.rs:27:5: 27:26 warning: unused import, #[warn(unused_imports)] on by default
src/drivers/lcd/c12332.rs:27 use core::slice::SliceExt;
```

But 2016-09-10 breaks with:
```
rustup run nightly-2016-06-10 cargo build --verbose --features mcu_k20 2>&1 | sed 's/.Users.nem/Home/g'
   Compiling ioreg v0.1.0 (file://Home/projects/lrn2rst/zinc/ioreg)
     Running `rustc ioreg/src/lib.rs --crate-name ioreg --crate-type dylib -C prefer-dynamic -g -C metadata=4302d6378f0cb21c -C extra-filename=-4302d6378f0cb21c --out-dir Home/projects/lrn2rst/zinc/target/debug/deps --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps`
   Compiling volatile_cell v1.0.0 (file://Home/projects/lrn2rst/zinc/volatile_cell)
     Running `rustc volatile_cell/lib.rs --crate-name volatile_cell --crate-type rlib -g -C metadata=6eaf9a8b4cfc05df -C extra-filename=-6eaf9a8b4cfc05df --out-dir Home/projects/lrn2rst/zinc/target/debug/deps --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps`
   Compiling zinc v0.1.0 (file://Home/projects/lrn2rst/zinc)
     Running `rustc build.rs --crate-name build_script_build --crate-type bin -g --cfg feature=\"mcu_k20\" --cfg feature=\"cpu_cortex-m4\" --out-dir Home/projects/lrn2rst/zinc/target/debug/build/zinc-d3eb462b2a6ccfcc --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps`
   Compiling rlibc v0.1.4 (https://github.com/hackndev/rlibc?branch=zinc#620489f8)
     Running `rustc Home/.cargo/git/checkouts/rlibc-22ab3e9e80121e06/zinc/src/lib.rs --crate-name rlibc --crate-type lib -g -C metadata=961f6a1dbd8e6b29 -C extra-filename=-961f6a1dbd8e6b29 --out-dir Home/projects/lrn2rst/zinc/target/debug/deps --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps --cap-lints allow`
     Running `Home/projects/lrn2rst/zinc/target/debug/build/zinc-d3eb462b2a6ccfcc/build-script-build`
ioreg/src/lib.rs:357:9: 357:28 error: cannot infer an appropriate lifetime for lifetime parameter `'a` due to conflicting requirements [E0495]
ioreg/src/lib.rs:357   match parser::Parser::new(cx, tts).parse_ioregs() {
                             ^~~~~~~~~~~~~~~~~~~
ioreg/src/lib.rs:360:27: 360:37 error: cannot infer an appropriate lifetime for lifetime parameter 'a in function call due to conflicting requirements [E0495]
ioreg/src/lib.rs:360       let items = builder.emit_items(cx, group);
                                               ^~~~~~~~~~
ioreg/src/lib.rs:355:1: 367:2 help: consider using an explicit lifetime parameter as shown: fn macro_ioregs<'a>(cx: &'a mut ExtCtxt<'a>, _: Span, tts: &[ast::TokenTree])
 -> Box<MacResult+ 'static>
ioreg/src/lib.rs:355 pub fn macro_ioregs(cx: &mut ExtCtxt, _: Span, tts: &[ast::TokenTree])
                     ^
ioreg/src/lib.rs:371:9: 371:28 error: cannot infer an appropriate lifetime for lifetime parameter `'a` due to conflicting requirements [E0495]
ioreg/src/lib.rs:371   match parser::Parser::new(cx, tts).parse_ioregs() {
                             ^~~~~~~~~~~~~~~~~~~
ioreg/src/lib.rs:374:27: 374:37 error: cannot infer an appropriate lifetime for lifetime parameter 'a in function call due to conflicting requirements [E0495]
ioreg/src/lib.rs:374       let items = builder.emit_items(cx, group);
                                               ^~~~~~~~~~
ioreg/src/lib.rs:369:1: 384:2 help: consider using an explicit lifetime parameter as shown: fn macro_ioregs_debug<'a>(cx: &'a mut ExtCtxt<'a>, _: Span,
                          tts: &[ast::TokenTree]) -> Box<MacResult+ 'static>
ioreg/src/lib.rs:369 pub fn macro_ioregs_debug(cx: &mut ExtCtxt, _: Span, tts: &[ast::TokenTree])
                     ^
error: aborting due to 4 previous errors
error: Could not compile `ioreg`.

Caused by:
  Process didn't exit successfully: `rustc ioreg/src/lib.rs --crate-name ioreg --crate-type dylib -C prefer-dynamic -g -C metadata=4302d6378f0cb21c -C extra-filename=-4302d6378f0cb21c --out-dir Home/projects/lrn2rst/zinc/target/debug/deps --emit=dep-info,link -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps -L dependency=Home/projects/lrn2rst/zinc/target/debug/deps` (exit code: 101)
```

This level of lifetime magic is a bit beyond me right now - I tried following the compiler's advice but got nowhere.

archaelus, 2016-06-24:
To be more precise: rustc 1.11.0-nightly (34505e222 2016-06-08) works, but rustc 1.11.0-nightly (7d2f75a95 2016-06-09) doesn't.

Phaiax, 2016-09-25:
fixed in master

mcoffin, 2016-11-29:
> fixed in master
Correct!

closed 2016-11-29 by mcoffin