hackndev/zinc #386: How to add support to new boards?

tengyifei, 2016-08-04:
Interested in adding one for Teensy (`MK20DX256`) and for STM Discovery (`STM32F746NG`).

farcaller, 2016-08-06:
The project is in a very big mess for a very long time since I can't dedicate enough personal time to it. There are a few pain points:

* ioregs requires constant maintenance as rust nightlies update
* platformtree is too fragile

The way to progress as I see it is to use boxed traits on stack, provided we reserve enough bytes for the magic. I've seen that done in other rust projects and it sounds like a good way to solve the issues I've faced in PT design. That will allow to use dynamic trait dispatch for HAL layer (not as effective as I'd want it to be, but simpler to move forward).

So, in regards to supporting new boards—there's little that could be done right now. There are no common traits between the platforms.

If you want to experiment—I'm ok with PRs that add functional traits for gpio, uart, spi that would work between platforms. You will need to use the new ioreg and define the full platform ioregs, and at least provide the initialisation sequence.

mcoffin, 2016-11-29:
I'm going to close this as the question has been answered. Feel free to re-open if you're still needing help, but I'm trying to clean up our waffle.io board.

closed 2016-11-29 by mcoffin