hackndev/zinc #387: example of new zinc project seem broken

morriswinkler, 2016-08-10:
Build seems to be broken.

i used example from https://github.com/posborne/zinc-example-lpc1768
with toolchain nightly-2016-05-26-x86_64-apple-darwin

here are my results
```
██ cargo build --verbose --release --target=thumbv7m-none-eabi
    Updating git repository `https://github.com/hackndev/zinc.git`
    Updating git repository `https://github.com/hackndev/rust-libcore`
error: no matching package named `core` found (required by `blink`)
location searched: https://github.com/hackndev/rust-libcore
version required: *
```

```
██ rustc --version
rustc 1.10.0-nightly (267cde259 2016-05-25)
```

```
██ cat Cargo.toml
[package]
name = "blink"
version = "0.1.0"
authors = ["Morris Winkler <m.winkler@laydrop.com>"]

[dependencies.zinc]
git = "https://github.com/hackndev/zinc.git"
features = ["mcu_stm32l1"]

[dependencies.macro_zinc]
git = "https://github.com/hackndev/zinc.git"
path = "macro_zinc"

[dependencies.macro_platformtree]
git = "https://github.com/hackndev/zinc.git"
path = "macro_platformtree"

[dependencies.ioreg]
git = "https://github.com/hackndev/zinc.git"
path = "ioreg"

[dependencies.platformtree]
git = "https://github.com/hackndev/zinc.git"
path = "platformtree"

[dependencies.core]
git = "https://github.com/hackndev/rust-libcore"

[[bin]]
name = "blink"
path = "src/blink.rs"
```

```
██ cat .cargo/config
[target.thumbv7em-none-eabi]
linker = "arm-none-eabi-gcc"
ar = "arm-none-eabi-ar"

[target.thumbv7m-none-eabi]
linker = "arm-none-eabi-gcc"
ar = "arm-none-eabi-ar"
```

```
██ cat thumbv7m-none-eabi.json
{
  "arch": "arm",
  "cpu": "cortex-m3",
  "data-layout": "e-m:e-p:32:32-i1:8:32-i8:8:32-i16:16:32-i64:64-v128:64:128-a:0:32-n32-S64",
  "disable-redzone": true,
  "executables": true,
  "llvm-target": "thumbv7m-none-eabi",
  "morestack": false,
  "os": "none",
  "relocation-model": "static",
  "target-endian": "little",
  "target-pointer-width": "32",
  "no-compiler-rt": true,
  "pre-link-args": [
    "-lm", "-lgcc", "-mcpu=cortex-m3", "-mthumb",
    "-Tlayout.ld"
  ]
}
```

---

I tried also latest nightly

```
██ cargo build --verbose --release --target=thumbv7m-none-eabi
warning: dependency (platformtree) specification is ambiguous. Only one of `git` or `path` is allowed. This will be considered an error in future versions
warning: dependency (macro_zinc) specification is ambiguous. Only one of `git` or `path` is allowed. This will be considered an error in future versions
warning: dependency (ioreg) specification is ambiguous. Only one of `git` or `path` is allowed. This will be considered an error in future versions
warning: dependency (macro_platformtree) specification is ambiguous. Only one of `git` or `path` is allowed. This will be considered an error in future versions
    Updating git repository `https://github.com/hackndev/zinc.git`
    Updating git repository `https://github.com/hackndev/rust-libcore`
error: no matching package named `core` found (required by `blink`)
location searched: https://github.com/hackndev/rust-libcore
version required: *
```

```
██ rustc --version
rustc 1.12.0-nightly (576f76659 2016-08-09)
```

posborne, 2016-08-10:
Hi @morriswinkler, I'll admit I haven't been working with Zinc much recently.  Things *should* work, but the project hasn't been too active recently (it would be nice to get this to change).

morriswinkler, 2016-08-13:
It seems to me since nightly is still the testbed for new features, and zinc relies on nightly it is hard to keep up with changes. Isn't it possible to go with one of the current stale releases of rust ?

farcaller, 2016-08-13:
That's the grand plan — get rid of compiler plugins and fall back to stable, otherwise it's too hard to maintain.

morriswinkler, 2016-08-14:
is that even possible ?

What are the steps necessary for that.

jaredr, 2016-08-27:
It looks like at least some plugins will be workable soon-ish on stable. See https://github.com/rust-lang/rfcs/pull/1681 .  Maybe this is enough for zinc's plugins to function?

gyscos, 2016-10-02:
Macro 1.1 looks like it will only support custom Derive; I think zinc would require something closer to what they call "Macro 2.0", which is still a long way off. :(

farcaller, 2016-10-02:
The plan is to pre-compile ioregs and move PT to the build script, so that there are no unstable macro requirements in the immediate app dependencies.

mcoffin, 2016-11-29:
Well, this *particular* error is fixed by the landing of #400, which deprecates the use of `rust-libcore`, in favor of letting [`xargo`](https://crates.io/crates/xargo) manage the potential compilation of the `core` crate for us.

closed 2016-11-29 by mcoffin