hackndev/zinc #395: Build blink_k20 failed

nvl1109, 2016-09-22:
I build blink_k20 with following failed, please take a look.

`$ cargo build --target=thumbv7em-none-eabi --features mcu_k20 --release
    Updating registry `https://github.com/rust-lang/crates.io-index`
    Updating git repository `https://github.com/hackndev/rlibc`
 Downloading rust-libcore v0.0.3
   Compiling macro_zinc v0.1.0 (file:///home/linhnv1/p/zinc/macro_zinc)
   Compiling zinc v0.1.0 (file:///home/linhnv1/p/zinc)
   Compiling ioreg v0.1.0 (file:///home/linhnv1/p/zinc/ioreg)
   Compiling rust-libcore v0.0.3
error: couldn't read "/home/linhnv1/p/zinc/macro_zinc/src/macro_zinc.rs": No such file or directory (os error 2)
Build failed, waiting for other jobs to finish...
error: Could not compile `macro_zinc`.

To learn more, run the command again with --verbose.`

`$ rustc --version
rustc 1.10.0-nightly (57ef01513 2016-05-23)`

`$ git log -2
commit 1ad41f3d21d602ca49cb2ff628c7c46dc2d714ff
Merge: 3a3ab46 f4a72b1
Author: Vladimir Pouzanov <farcaller@gmail.com>
Date:   Sun Sep 18 11:50:24 2016 +0100

    Merge pull request #392 from rnestler/improve_readme

    Add a note about which rust version to use`

Thanks.

Phaiax, 2016-09-22:
Hm, just checked out 1ad41f3 and compiled with that same rustc and it worked.
Have you moved any folders or changed anything? git status -s

dhylands, 2016-09-23:
I also just tried a fresh clone of zinc (same hash as @Phaiax mentioned) and it worked (also using the same version of rustc). I was running on Ubuntu 16.04 (if that matters).

Phaiax, 2016-09-23:
note that the compatible nightly version just has changed

nvl1109, 2016-09-24:
Hi all,

Firstly, I'm so sorry for late response. My environment is not REAL Ubuntu 14.04, it is Windows 10 Subsytem Ubuntu. Although it still is Ubuntu 14.04 but maybe has some different.

I'm do not have a chance to try in REAL Linux (I have archlinux dual boot, but I'm busy for some days). I will try then feedback later.

Thank you guys for your responses.

nvl1109, 2016-09-26:
@Phaiax I'm tested on Archlinux with updated nightly rust version. It's ok. So I will close this issue. Please accept my apology.

closed 2016-09-26 by nvl1109

Phaiax, 2016-09-26:
Hm, no need to apologize :D
`/home/linhnv1/p/zinc/macro_zinc/src/macro_zinc.rs`
I'm curious why it wanted to access this file, it did not even exist.