hackndev/zinc #397: Example `blink_tiva_c` does not compile

gyscos, 2016-09-24:
When running the recommended nightly version (`rustc --version` returns `d4037fc47 2016-09-16`), the `examples/blink_tiva_c` example does not compile for two reasons:
* `#![deny(missing_docs)]` errors because of the missing doc in the tiva_c module (See #373)
* Even after fixing this, I get:
```
% cargo build --target=thumbv7em-none-eabi --features mcu_tiva_c --release
   Compiling zinc v0.1.0 (file:///home/gyscos/fetched/zinc)
warning: unused or unknown feature, #[warn(unused_features)] on by default
  --> /home/gyscos/fetched/zinc/src/lib.rs:17:45
   |
17 | #![feature(core_intrinsics, core_slice_ext, core_str_ext)]
   |                                             ^^^^^^^^^^^^

warning: unused import, #[warn(unused_imports)] on by default
  --> /home/gyscos/fetched/zinc/src/drivers/lcd/c12332.rs:27:5
   |
27 | use core::slice::SliceExt;
   |     ^^^^^^^^^^^^^^^^^^^^^

   Compiling blink_tiva_c v0.0.1 (file:///home/gyscos/fetched/zinc/examples/blink_tiva_c)
error[E0433]: failed to resolve. Use of undeclared type or module `pt`
  --> src/main.rs:45:15
   |
45 | fn run(args: &pt::run_args) {
   |               ^^^^^^^^^^^^ Use of undeclared type or module `pt`

error: aborting due to previous error

error: Could not compile `blink_tiva_c`.

To learn more, run the command again with --verbose.
cargo build --target=thumbv7em-none-eabi --features mcu_tiva_c --release  11.20s user 0.10s system 99% cpu 11.338 total
```

jamwaffles, 2016-10-12:
I ran into the same problems and eventually managed to fix them. The changes were merged in #399. I managed to compile the Tiva examples with Rust 1.13.0-nightly. There's a sample project [here](https://github.com/jamwaffles/zinc-tiva-c-example) that you could try and build on. It seems Zinc requires quite a specific version of Rust to work properly at the moment. YMMV of course :)

mcoffin, 2016-11-29:
Confirmed that this example builds as of e68f861, so I'm going to close this. I'm leaving #373 open since #399 just basically ignored-away the `missing_docs` errors, and we really should go back through and fix that instead of just ignoring the errors / warnings.

I noted that on #373 as well.

closed 2016-11-29 by mcoffin