hackndev/zinc #398: NGX LPC1768 Blueboard with USB bootloader

jamwaffles, 2016-10-08:
Apologies for putting a question in a Github issue, but I can't think of a better place to ask it.

I've been trying to compile a hello world LED blink test program for the Blueboard mentioned in the title. It uses an LPC1768-H IC, however it also has a USB bootloader that shows up as a mass storage device, which I believe to be the [AN10866](http://www.nxp.com/documents/application_note/AN10866.pdf) secondary bootloader.

With this bootloader, program code should be offset by `0x2000`, but I'm having trouble getting the linker to do this, as well as generate code in the correct binary format. I have a sample project compiled with Keil uVision 5 which works well, but doing `file` on the binary tells me it's a COFF file. When I build Zinc (and do `objcopy`), I get a different format reported with a wildly different layout when viewed in a hex editor. I've scoured the net in attempts to get `objcopy` to output a COFF file to no avail.

Save for pasting an avalanche of code and attachments, please let me know what further information you need to at least put me on the right track. I've been walking circles around various `.ld` files all day with no progress and could really use some help!

mcoffin, 2016-11-29:
The IRC channel (`#zinc` on the Mozilla network) has been seeing a little more activity as of late. I'm not entirely sure what you're looking for, but if you hop over there and highlight me I'd be more than happy to help fix it with ya.

jamwaffles, 2016-11-29:
Thanks, I'll drop in there if I have something else come up. I ended up giving up on the LPC board due to the _vast_ lack of documentation for it and instead moved on to the Tiva C I had sitting in my drawer, which works pretty well with Zinc.

closed 2016-11-29 by jamwaffles