hackndev/zinc #410: Add support for Espressif ESP32

darkdragon-001, 2017-02-27:
This interest for [Espressiv ESP32](https://en.wikipedia.org/wiki/ESP32) is growing really fast at the moment, especially because of their big functionality on a small footprint for a very low price. 

I would love to see this supported :smiley:

farcaller, 2017-02-27:
I believe there's no llvm support, thus it's not possible to target the cpu from rust. @whitequark might know more.

Still closing, as it's not something zinc can technically do.

closed 2017-02-27 by farcaller

whitequark, 2017-02-27:
The LLVM backend is reportedly being developed but not by me. I don't know what's its status other than "Espressif is probably paying for it".

darkdragon-001, 2017-02-28:
Who is developing it? Where can I keep track of the progress?

whitequark, 2017-02-28:
Some consulting company. I don't think you can.

darkdragon-001, 2017-02-28:
Thanks for the clarification...

@whitequark I found the [public part of your discussion](https://www.esp32.com/viewtopic.php?t=499) as well as the [official confirmation tweet](https://twitter.com/EspressifSystem/status/824399990965489664).

I also found some files in the [llvm project](https://llvm.org/svn/llvm-project/llvm-gcc-4.2/trunk/gcc/config/xtensa/).

If anybody stumbles over an upgrade regarding this topic, please keep me informed!

whitequark, 2017-02-28:
> I also found some files in the llvm project.

This is just from a copy of gcc inside llvm. It has no bearing on llvm itself.