hackndev/zinc #411: rust-src component not found on MacOS

MrSpock, 2017-03-21:
Hi,
When I try build basic example on MacOS I'm getting message about missing rust-src which is installed:
spock@macspock:~/zinc/examples/blink_tiva_c> xargo build --target=thumbv7em-none-eabi --features mcu_tiva_c --release
Error(Msg("`rust-src` component not found. Run `rustup component add rust-src`."), (None, None))
spock@macspock:~/zinc/examples/blink_tiva_c> rustup component add rust-src
info: component 'rust-src' is up to date
spock@macspock:~/zinc/examples/blink_tiva_c> uname -a
Darwin macspock 16.4.0 Darwin Kernel Version 16.4.0: Thu Dec 22 22:53:21 PST 2016; root:xnu-3789.41.3~3/RELEASE_X86_64 x86_64
spock@macspock:~/zinc/examples/blink_tiva_c>

mcoffin, 2017-03-21:
Hmm, likely `rustup` is trying to install `rust-src` for a different toolchain than the one you have overridden for the zinc directory. If you have IRC, I'll be chilling in #zinc on Mozilla.

MrSpock, 2017-03-22:
Hi.
Source of this problem was in xargo (https://github.com/japaric/xargo/issues/85).
Update for xargo and problem is solved.

closed 2017-03-22 by mcoffin