hackndev/zinc #412: Example blink_k20_isr from README doens't build

MrSpock, 2017-05-22:
Hi,
When trying to build basic example here is what is happening:
```
xargo build --target=thumbv7em-none-eabi --features mcu_k20 --release
   Compiling core v0.0.0 (file:///Users/spock/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/src/rust/src/libcore)
    Finished release [optimized] target(s) in 15.22 secs
    Updating registry `https://github.com/rust-lang/crates.io-index`
 Downloading rlibc v1.0.0
   Compiling ioreg v0.1.0 (file:///Users/spock/zinc/ioreg)
   Compiling zinc v0.1.0 (file:///Users/spock/zinc)
   Compiling volatile_cell v1.0.0 (file:///Users/spock/zinc/volatile_cell)
   Compiling rlibc v1.0.0
error[E0432]: unresolved import `syntax::codemap::mk_sp`
  --> /Users/spock/zinc/ioreg/src/parser.rs:21:61
   |
21 | use syntax::codemap::{Span, Spanned, respan, dummy_spanned, mk_sp};
   |                                                             ^^^^^ no `mk_sp` in `codemap`

error[E0432]: unresolved import `syntax::parse::lexer::Reader`
  --> /Users/spock/zinc/ioreg/src/parser.rs:26:5
   |
26 | use syntax::parse::lexer::Reader;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `Reader` in `parse::lexer`

error[E0432]: unresolved import `syntax::codemap::mk_sp`
  --> /Users/spock/zinc/ioreg/src/builder/register.rs:21:31
   |
21 | use syntax::codemap::{respan, mk_sp};
   |                               ^^^^^ no `mk_sp` in `codemap`

error[E0412]: cannot find type `TtReader` in module `lexer`
  --> /Users/spock/zinc/ioreg/src/parser.rs:44:18
   |
44 |   reader: lexer::TtReader<'a>,
   |                  ^^^^^^^^ not found in `lexer`

error[E0425]: cannot find function `new_tt_reader` in module `lexer`
  --> /Users/spock/zinc/ioreg/src/parser.rs:56:29
   |
56 |     let mut reader = lexer::new_tt_reader(&sess.span_diagnostic, None, None, ttsvec);
   |                             ^^^^^^^^^^^^^ not found in `lexer`

error[E0433]: failed to resolve. Could not find `InternedString` in `token`
  --> /Users/spock/zinc/ioreg/src/builder/utils.rs:54:50
   |
54 |    .map(|word| cx.meta_list_item_word(word.span, token::InternedString::new(word.node)));
   |                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^ Could not find `InternedString` in `token`

error[E0433]: failed to resolve. Could not find `InternedString` in `token`
  --> /Users/spock/zinc/ioreg/src/builder/utils.rs:55:39
   |
55 |   let allow = cx.meta_list(name.span, token::InternedString::new(name.node),
   |                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^ Could not find `InternedString` in `token`

error[E0412]: cannot find type `InternedString` in module `token`
  --> /Users/spock/zinc/ioreg/src/builder/utils.rs:61:54
   |
61 | pub fn doc_attribute(cx: &ExtCtxt, docstring: token::InternedString)
   |                                                      ^^^^^^^^^^^^^^ not found in `token`
   |
help: possible candidate is found in another module, you can import it into scope
   | use syntax::symbol::InternedString;

error[E0433]: failed to resolve. Could not find `InternedString` in `token`
  --> /Users/spock/zinc/ioreg/src/builder/utils.rs:67:34
   |
67 |     cx.meta_name_value(DUMMY_SP, token::InternedString::new("doc"), s);
   |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^ Could not find `InternedString` in `token`

error[E0412]: cannot find type `InternedString` in module `token`
   --> /Users/spock/zinc/ioreg/src/builder/utils.rs:178:57
    |
178 | pub fn intern_string(cx: &ExtCtxt, s: String) -> token::InternedString {
    |                                                         ^^^^^^^^^^^^^^ not found in `token`
    |
help: possible candidate is found in another module, you can import it into scope
    | use syntax::symbol::InternedString;

error[E0061]: this function takes 3 parameters but 4 parameters were supplied
   --> /Users/spock/zinc/ioreg/src/parser.rs:545:38
    |
545 |         let lit = parse::integer_lit(n.as_str().deref(),
    |                                      ^^^^^^^^^^^^^^^^^^ expected 3 parameters

error[E0308]: mismatched types
   --> /Users/spock/zinc/ioreg/src/parser.rs:550:11
    |
550 |           ast::LitKind::Int(n, _)  => Some(n),
    |           ^^^^^^^^^^^^^^^^^^^^^^^ expected enum `std::option::Option`, found enum `syntax::ast::LitKind`
    |
    = note: expected type `std::option::Option<syntax::ast::LitKind>`
               found type `syntax::ast::LitKind`

error[E0308]: mismatched types
   --> /Users/spock/zinc/ioreg/src/parser.rs:550:44
    |
550 |           ast::LitKind::Int(n, _)  => Some(n),
    |                                            ^ expected u64, found u128

error[E0308]: mismatched types
  --> /Users/spock/zinc/ioreg/src/builder/utils.rs:29:41
   |
29 |   cx.expr_lit(n.span, ast::LitKind::Int(n.node, ast::LitIntType::Unsuffixed))
   |                                         ^^^^^^ expected u128, found u64

error[E0023]: this pattern has 6 fields, but the corresponding tuple variant has 7 fields
   --> /Users/spock/zinc/ioreg/src/builder/utils.rs:136:5
    |
136 |     ast::ItemKind::Impl(_, _, _, _, _, ref items) => {
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected 7 fields, found 6

error[E0308]: mismatched types
   --> /Users/spock/zinc/ioreg/src/builder/getter.rs:140:29
    |
140 |           ast::LitKind::Int(val, ast::LitIntType::Unsuffixed));
    |                             ^^^ expected u128, found u64

error[E0308]: mismatched types
  --> /Users/spock/zinc/ioreg/src/builder/union.rs:88:41
   |
88 |   cx.expr_lit(n.span, ast::LitKind::Int(n.node as u64, ast::LitIntType::Unsigned(ast::UintTy::Us)))
   |                                         ^^^^^^^^^^^^^ expected u128, found u64

error: no associated item named `FixedLengthVec` found for type `syntax::ast::TyKind` in the current scope
   --> /Users/spock/zinc/ioreg/src/builder/union.rs:101:13
    |
101 |             ast::TyKind::FixedLengthVec(base_ty, expr_usize(cx, respan(reg.count.span, n as u64)))),
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: no associated item named `FixedLengthVec` found for type `syntax::ast::TyKind` in the current scope
   --> /Users/spock/zinc/ioreg/src/builder/union.rs:149:13
    |
149 |             ast::TyKind::FixedLengthVec(u8_ty, expr_usize(self.cx, respan(DUMMY_SP, length))));
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to 9 previous errors

error: Could not compile `ioreg`.
```

mcoffin, 2017-05-22:
Are you using the recommended old nightly version of rust?

MrSpock, 2017-05-22:
Argh - I assumed that every newer nightly or stable will have required features.
My mistake.

mcoffin, 2017-05-30:
It's not necessarily that they got rid of features, it's just that compiler internals change a lot right now, and this kind of compiler plugin has to change along with it, but there's nobody that's putting in the work right now to keep it up to date.

closed 2017-05-30 by mcoffin