hackndev/zinc #416: [Architecture] RISC-V HiFive1 Support?

kallisti5, 2018-06-06:
I know it's early days (and other MCU support is coming), but the HiFive1 seems like an *awesome* potential target platform for something like this.

At the moment there is a gcc / g++ toolchain you can build binaries for RISC-V with, and load them onto the HiFive1 with openocd.

https://www.sifive.com/products/hifive1/

jamwaffles, 2018-06-06:
Last I heard Zinc was abandoned which is a shame. Have you looked at [rust-lang-nursery/embedded-wg](https://github.com/rust-lang-nursery/embedded-wg)? I mostly live in the STM32 world at the moment, but I think there are murmurings of Rust RISC-V support over there. I'd love to see Rust as a first class citizen on RISC-V so make some noise!

johnthagen, 2018-07-13:
> Last I heard Zinc was abandoned 

Yes, https://users.rust-lang.org/t/zinc-mini-post-mortem/7079

johnthagen, 2018-07-17:
@kallisti5 Check out https://github.com/rust-lang-nursery/embedded-wg for actively maintained Rust embedded projects.